collin80/due_can #8: Communication problem between Arduinos

DavideBisio, 2015-02-24:
I need to make two Arduino boards communicate. I've written this code to send messages:
...
void setup() {
  etc..
  Can0.begin(CAN_BAUD_RATE);
  etc..
}

void loop() {
  CAN_FRAME canFrame;
  uint16_t varA, varB, varC;
  
  etc..
  
  canFrame.extended = false;
  canFrame.id = MY_ID;
  canFrame.length = 6;
  
  canFrame.data.byte[0] = (varA & 0xFF00) >> 8;
  canFrame.data.byte[1] = (varA & 0xFF);
  
  canFrame.data.byte[2] = (varB & 0xFF00) >> 8;
  canFrame.data.byte[3] = (varB & 0xFF);
  
  canFrame.data.byte[4] = (varC & 0xFF00) >> 8;
  canFrame.data.byte[5] = (varC & 0xFF);

CAN.sendFrame(canFrame);
}
...
I can read messages using a CAN-bus analyzer with correct data and timing.

To read messages I've written another program, loaded on another Arduino, connected to the first:
...
void setup() {
  etc..
  Can0.begin(CAN_BAUD_RATE);
  for (int i=0; i<7; i++) Can0.setRXFilter(i, 0, 0, false);
  etc..
}

void loop() {
  if (Can0.available()) {
    CAN_FRAME incomingCanFrame;
    Can0.read(incomingCanFrame);
    // Code to use message's data
  }
}
...

This Arduino can't read data, even if it is correctly sent by the first arduino and received by the CAN-bus analyzer.
The thing that most I can not understand is that if I copy the message received by the CAN-bus analyzer and I send it back, Arduino can receive it and it does everything it has to do.

Can you help me to solve this problem?

collin80, 2015-02-24:
Hmm, are you sure that you have proper bus termination? It's possible that improper termination could do this. Make sure you have 120 ohms of resistance on both ends of the bus. You should get a value of 60 ohms if you measure from CANH to CANL with everything off.  The fact that both ends work with the canbus analyzer suggests that you are very close to getting it to work.

DavideBisio, 2015-02-25:
Thank you!
It is my first experience with Arduino and CAN-bus, and I got stuck because I thought it was a software problem, but you confirmed me those two lines of code are good.
I cecked impedance all over the bus: from CANH to CANL, from CANH to GND and from CANL to GND. Everything was ok, but when I checked signal with my oscilloscope I realized that a CAN transceiver wasn't well soldered: it couldn't reach CANL wire.
I'm surprised that the can analyzer was able to read the signal, while lacking CANL wire!

closed 2015-02-25 by DavideBisio