collin80/due_can #9: current library in master fails to transmit

nostar, 2015-03-11:
Testing the current master branch and it receives fine but fails to transmit using the following code:

int send_frame()
{
  int r;
  CAN_FRAME tx;
  tx.id = 0x100;
  tx.length = 4;
  tx.extended = 0;
  tx.data.byte[0] = 0x01;
  tx.data.byte[1] = 0x02;
  tx.data.byte[2] = 0x03;
  tx.data.byte[3] = 0x04;
  r = Can0.sendFrame(tx);
  Serial.print("sent tx r = ");
  Serial.print(r);
  Serial.print("\n");
  return r;
}

It returns true for 10 or so attempts then returns false and no frames make it to the bus.  Also receive appears to start missing frames every time i attempt to transmit.  I am using the CAN_TrafficSnooper example and just added this send_frame() function.  I tried sending a couple different ways like on a 500ms SimpleTimer and just a 1 shot after receipt of a given frame for example.  Before I start digging deeper and troubleshooting the library further I wanted to see if this is still a work in progress and is this already known behavior.

collin80, 2015-03-11:
It's most certainly not known behavior. I suppose I'd call the library a work in progress as there are still things that would be nice to do. However, it really ought to work. I use it all of the time and so do many other people. So, I'm reasonably sure that the library is currently functional. This leads me to believe that something else must be going on - termination issues, bad hardware, something wrong with the version of the Arduino IDE you are using, etc.

nostar, 2015-03-12:
OK I forgot how to count.  I had pin 5 (Vref) tied to ground instead of pin 8 (Rs).  It works great now, thanks!

closed 2015-10-08 by collin80