collin80/due_can #10: watchFor()

Tdarty, 2015-05-06:
Hi,
I use CAN_EchoTest.ino in exemple.
I try to use "Can0.watchFor();".
It does'nt work without id. (no interupt for the mailbox)
When i use "Can0.watchFor(0x007);" i recive my frame correctly.

I look in the due_can.cpp and i see :

int CANRaw::watchFor() 
{
	int retVal = setRXFilter(0, 0, false);
	enable_interrupt(getMailboxIer(0));
	setRXFilter(0, 0, true);
	enable_interrupt(getMailboxIer(1));

	return retVal;
}

I don't know if it's normal but i don't understand how it work.

collin80, 2015-05-06:
I've had another report of this so I'm not going to say that you're wrong here. For some reason the watchFor() function does seem to be broken. I will keep this issue open and try to get to the bottom of what is going on.

collin80, 2015-05-12:
This problem should now be corrected in the most recent version. Closing the ticket.

closed 2015-05-12 by collin80