collin80/due_can #13: How to use this library

Jeshwanth, 2015-12-09:
Hello @collin80 ,

We are looking for using due_can library for our project, We have Arduino due and we are new to CAN protocol.
We need some clarity on how to use your library .
1. Is there any hardware dependency ? (does it need seperate transceiver etc ?)
2. Can OBD2 be directly connected to Due ? Are these compatible ?

Thanks in Advance

hannes93, 2015-12-11:
Hey,
I just use the library but I think you can do a lot of things with it.
1. As there is no transceiver on the Due you need some kind of breakout board (CAN shield). I use the board below:
http://skpang.co.uk/catalog/canbus-breakout-board-33v-teensy-31-compatible-p-1396.html
(just be sure that it runs with 3,3V operating speed)
or you can buy the transceiver by yourself and build your own breakout board
2. For example the board above comes with an OBD interface: I am not sure what is exactly needed but I think if you want to to build an OBD interface by your own you need some resistors for the CAN bus at least.

collin80, 2015-12-11:
Yeah, sorry, I forgot to comment on this.

1. You do need transceiver chips. The Due has the canbus hardware other than transceivers and termination resistors. If your canbus is already terminated then all you need are the transceivers and attendant hardware for the transceiver like resistors and a connector.

2. Yes, modern vehicles 2008 and newer will have canbus on the OBDII port. Some older vehicles will as well but the newer, the greater chance. You will obviously need an adapter to go from OBDII to the transceiver. 

If you are interested in OBDII connection then you can find some of that stuff here: http://store.evtv.me/products.php?cat=28

You won't find a shield on that site that accepts the DB9 plug on the end of the OBDII cord but EVTV used to sell a shield that has the proper plug so you can ask if they've still got some and just don't have it on the site anymore.

Alchemy455, 2018-06-14:
It is mentioned in the readme that I also need to use the can_common library.  When I include that library the compiler tells me there are multiple definitions of many items.  Any advice on implementation using both the due_can and can_common libraries?  Is the readme wrong? because my software builds with no problems when I only include the due_can library. However, using just the due_can library with 2 transceivers on both the due CAN channels I can not get an echo test to work. I've read through the ISO11898 (CAN buss standard) and everything seems to be correct electrically. I'm using the Texas Instruments SN65HVD23x 3.3V CAN Bus Transceivers. Thanks for any help.

redbaroon, 2023-07-24:
> 

hello, some fix to the problem of multiple definitions? , 2023 and im trapped on same 🍰

collin80, 2023-07-25:
Update both can_common and this library. The newest versions of each don't duplicate definitions. You will then have to make sure there aren't any old copies lying around that the IDE could pick up and use.