collin80/due_can #14: Quick data exchange

hannes93, 2015-12-11:
Hey collin,
I am trying to receive 4 standard CAN frames, which are sent by an ECU, and respond to them in time. The frames are sent in a cycle of 10 ms. I setup a mailbox for each frame (e.g. 0x110, 0x111, 0x112, 0x113) and tell the callback to respond with my own frames  (e.g. 0x210, 0x211, 0x212, 0x213) but when I look at the CAN log I can see that my frames are to slow to be transmitted before I get the next frames form the ECU. Is the system too slow or am I doing something wrong?

collin80, 2015-12-15:
I tried this out myself and it seems to work very rapidly. Here is my test setup (which was probably very different from your test setup): I used two EVTVDue boards connected together. I had both ends terminated. I put a kvaser leaf light in the middle tapping the CANH and CANL lines so I could monitor what was going on. Then I used the Can_SnooperCallback example included with the library as one end. I modified it such that instead of printing to the console it merely changes the received ID by adding one and resending the frame. On the other EVTVDue I put my GVRET sketch and used SavvyCAN to periodically send frames out. With the kvaser in the middle I was able to see both sides sending and measure how long it took for responses to happen. My result is that the CAN_SnooperCallback (modified) side would send the reply about 2-3 ten thousandths (0.0001) of a second after reception. That's pretty fast. Because of that I believe the library to be working properly.

So, the question is then, why is it working slowly for you. My best guess would be that you have debugging statements or some sort of serial or USB traffic going on. If you print things to Serial or SerialUSB you are likely to get severe pauses at seemingly random times. USB is polled and so you will find that sending too much will cause your sketch to block until the next time the USB is polled. This isn't long, perhaps 10ms but those 10ms could be a bad thing if you needed to reply sooner. Try running your code without any debugging or other sort of serial output and see what happens.

hannes93, 2015-12-15:
Your guess was correct. I used debugging to monitor the incoming messages over serial. When I remove the debugging statements I achieve a response within (0.0002 - 0.0003 sec). That's fast enough for me to achieve the things I want to do. Thanks a lot for your support!

closed 2015-12-15 by hannes93