collin80/due_can #21: Library Compatibility with CanOPEN

AlphaSmile, 2016-05-17:
Hello, I'd like to use your library in a school project on a variable speed drive(SEVCON Gen4) which is communicating with the CanOPEN protocol. Apparently, this protocol is much more elaborated than the      
the basic CAN one, and I wanted to know if you're library was designed to fit this particularity or not. Currently I'm trying to upgrade it by looking into another project which use an arduino and request some information from the SEVCON Gen4.
If you want to take a look, here it is : 
https://github.com/SuperTango/TangoLogger

collin80, 2016-05-17:
The base due_can library has no concept of CANOpen at all. However, I did
have a need to use canopen devices in the past and wrote some code for
that. In my github repos is also due_canopen which extends due_can with
functions you might want to use with canopen.

https://github.com/collin80/due_canopen

I can't say it is complete by any means but it might be helpful to you.

AlphaSmile, 2016-05-18:
Thanks for your answer. I'll have a look at your link, it might be what I'm looking for. I also found the CANopenNode project which is quite interesting regarding the adaptability of the library on differents platforms.

closed 2016-05-18 by AlphaSmile