collin80/due_can #22: Single Wire CAN functionality

arkraven000, 2016-10-19:
I'm working with the EVTV CanDue 2.0 shield. Documentation has lead me to your repo but I think I might be missing something. I haven't been able to find any functionality in your source code that mentions "swcan" or "single wire" for transmitting or receiving messages. Did I miss a file? If not, do you have a suggestion for where to go next?

Mike4U, 2016-10-22:
I don't have the EVTV CanDue 2.0 shield but have used due_can on a homemade shield with sw can.  Due_can library doesn't know or care whether its sw can or regular can.  Just set the correct speed and use the swcan interface on your board.  The documentation on your board at EVTV shows the enable pin required and other details if I recall.

arkraven000, 2016-10-24:
Thank you for clearing that up. I was suspicious that was the case.

collin80, 2016-10-26:
The only part of single wire CAN that is really different is that SWCan has
a mode where a high voltage wake up is used. This is a 12V signal down the
CAN wire and is used to wake up devices on the bus. If you don't need that
functionality then everything else is the same. If you do need that
functionality then the GVRET project has support for doing 12V wake up
across SWCAN.