collin80/due_can #23: Problems with DLC=1

moffenbach, 2017-01-19:
Hello,
I started a project with the due_can library. But I have some problems and want to ask for your help to solve them one after the other.
1. My project is small at the moment and works well, if I send a 8 byte long frame inside the main loop by the sendFrame method. Trying to send another frame with a DLC (datalength) of 1 byte, only the ID and the data length code will transmitted. Changing the DLC to 2 two bytes of data are sended. Do anybody have an idea?
2. I do not receive any frames. I initialized the library with Can0.begin(250000); ank check for frames by Can0.available() > 0 in a cycling task loop. Is it okay so far or is some initialisatin missing?

Thank's for you help
Michael

collin80, 2017-01-20:
1. I hadn't heard of any issues with DLC=1. There is no practical reason
why that number of data bytes should be special in any way. I'll have to
test this when I get a chance.

2. You need to set some filters to let traffic through. Here's a quick code
snippet:

//By default there are 7 mailboxes for each device that are RX boxes
//This sets each mailbox to have an open filter that will accept extended
//or standard frames
int filter;
//extended
for (filter = 0; filter < 3; filter++) {
Can0.setRXFilter(filter, 0, 0, true);
Can1.setRXFilter(filter, 0, 0, true);
}
//standard
for (filter = 3; filter < 7; filter++) {
Can0.setRXFilter(filter, 0, 0, false);
Can1.setRXFilter(filter, 0, 0, false);
}

Adjust accordingly for your own code, perhaps you don't have a Can1 on your
board.

moffenbach, 2017-01-20:
Hi Collin,
thanks for your help.
I don't know why, but the first problem disappears by itself (???).

About my second problem I'd like to say, that according the documentation, no filter settings are needed. Also in the ExtendedPingPong example there are no filter settings. So I thing there is a need to update the documentation.
Tomorrow I will try to insert the posted filter settings - I will report !

moffenbach, 2017-01-20:
Hi Collin,
I set the filters in the way you posted me and everything works well so far. If I will have other problems, please allow me to ask you for help again.

Best regards
Michael

closed 2017-01-20 by moffenbach