collin80/due_can #27: Error in DUE_2CAN_Channel_TEST Sketch file

GFMuayThai, 2017-09-08:
Hi, 
I've just downloaded the Due_2CAN_Chyannel_Test sketch and uploaded to my arduino 2.

After file compilation, the following error appeared:

Arduino:1.8.4 (Windows 7), Scheda:"Arduino/Genuino Uno"

In file included from sketch\DueCANLayer.h:4:0,

                 from sketch\DueCANLayer.cpp:4:

due_can.h:177: error: 'Can' does not name a type

  Can* m_pCan ;

  ^

due_can.h:195: error: expected ')' before '*' token

     CANRaw( Can* pCan, uint32_t En);

                ^

exit status 1
'Can' does not name a type

Someone could help me, please?

I think that the problem is with the setting of the variable type for "Can".

There could be some other issue like that?

THank You in advance
Bye

collin80, 2017-09-08:
You forgot to select Arduino Due as the board you are compiling for. I can see in your data this line:

Scheda:"Arduino/Genuino Uno"

That should reference the Due instead. If this is your first time compiling Due sketches you might need to use the board manager to download the Due support files first.

GFMuayThai, 2017-09-16:
OK.

Sorry but I'm not accustomed with Arduino.

Thank You very much.
Bye