collin80/due_can #28: How to use RTR with due_can library ?

mattiabalutto, 2017-09-14:
Hi, 

I'm working with your library and I will to send a RTR frame. In your library I found the CAN_FRAME struct definition, where I have found the rtr variable:

```
typedef struct {
uint32_t id;      // EID if ide set, SID otherwise
   uint32_t fid;     // family ID
   uint8_t rtr;      // Remote Transmission Request <-- THIS
   uint8_t priority; // Priority but only important for TX frames and then only for special uses.
   uint8_t extended; // Extended ID flag
   uint8_t length;      // Number of data bytes
   BytesUnion data;  // 64 bits - lots of ways to access it.
} CAN_FRAME;
```

I have tried to send a frame with this variable but nothing change. What's wrong ?

Sketch Code:
```
#include "variant.h"
#include <due_can.h>

bool initDevices = false;

void setup() {
  Can0.begin(CAN_BPS_500K);
}

void loop() {
  CAN_FRAME output;

  if(!initDevices) { 
    output.id = 0x21;
    output.extended = false;
    output.length = 1;
    output.data.byte[0] = 0x05;
    CAN.sendFrame(output);
    
    initDevices = true;
  }

  output.id = 0x321;
  output.extended = false;
  output.rtr = 0x01;
  output.length = 0;
  CAN.sendFrame(output);

  delay(50);
}

```

Thank You in advance
Bye

P.S. Sorry for my English

csBlueChip, 2018-03-05:
This library does NOT support RTR
The RTR variable is declared, but it is NEVER used.

I have patched the library to READ the RTR flag, https://github.com/collin80/due_can/issues/33
...but I have NOT (yet) patched it to WRITE the RTR flag

collin80, 2018-03-06:
Yeah, I really don't like the concept of RTR frames and I never did anything to support them. Maybe I kind of passively neglected to even implement RTR because I find it to be pointless. It has been deprecated for quite some time and isn't part of the CAN-FD standard at all. But, I suppose my dislike of RTR shouldn't mean that nobody else can use it if they wanted to. So, I really should merge your changes in. I'll try to get that done in the near future.

BJoubert89, 2021-02-22:
I am using the due_can library and also noticed that setting the RTR flag does nothing. RTR support would be great as it is part of the CAN standard and is supported by the Due. I don't find it useless at all. Anyway, thanks for the library.