collin80/due_can #29: master version does not work ok

smileage001, 2017-09-30:
Hi collin, 
    Thank you for your nice job. I download master version, it is not OK when I test the examples. Eg，
 CAN_EchoTest , can1 receives wrong messages. Even when I remove the transceivers which connect can0 and can1, it can still receive messages.
    So, I try the release version due_can-2.0.1, it works very well. I don't know the reason, and wonder why. 
    I use Due, programming port for test. Thanks.

collin80, 2017-10-01:
Well, that's kind of weird. There have been some big changes recently but I have been testing and I don't remember seeing an issue like this happen. I'll give it a try on my end and see what happens.

smileage001, 2017-10-05:
Hi collin80，
When I was debugging， I found “Variation=7.0; ” in BoatData.h, Which should be in radians here. 
Thank you for your enthusiastic reply.