collin80/due_can #39: Update library for CAN-FD?

tlewiscpp, 2019-02-24:
Is the Arduino Due theoretically capable of supporting [CAN FD](https://en.wikipedia.org/wiki/CAN_FD)? If so, would this library be able to be updated to support it? From the looks of it, CAN-FD is simply an extension that allows 64 data bytes per frame, rather than 8 data bytes. It looks to be compatible with existing CAN2.0 networks. If it is capable, I would be willing to help the programming/testing effort with a CAN-FD transceiver (maybe the MCP2562FD?).

collin80, 2019-02-25:
Nope, there is no hardware support for CAN-FD in the Atmel SAM3X processor line. I do have support for CAN-FD added to my ESP32-CAN library. That uses a SPI connected MCP2517FD chip. That same chip could be added to a Due compatible board to add CAN-FD support.

closed 2019-02-25 by collin80

ship23-95, 2020-07-27:
@collin80  Sir i also want to use CANFD please tell how to do .I have Rugged Board which have linux

collin80, 2020-07-29:
You need something that supports CAN-FD directly. It isn't as simple as just sending 64 bytes instead of 8. The speed of the bus increases during the sending of data bytes and there are other changes. So, the way forward is to use either a processor with CAN-FD (they do exist) or use an SPI connected chip like the MCP2518FD. With such a chip, and a compatible transceiver like the MCP2558FD you can then send and receive CAN-FD messages. There is a linux driver for the MCP2518FD so you're good there.