collin80/due_can #41: Due_can execute code in ram classe error

autoprog, 2019-08-01:
Hi,
I tested Due_can code in flash runnig ok.
now compile program for execute in RAM, cause error.

I tested IDE PIO and eclipse.

othes function with ISR ok, serial, TC2...

Problems " ", call classe  CANRaw, and write register in **m_pCan**

see my function test:
[image]

if write register error,
  **Can0.init( CAN_BPS_500K);
  Can0.watchFor();**
thanks,
Carlos.

collin80, 2019-08-01:
I have to admit that I have never tried to move the routines into RAM. I do not know why doing so would cause a hard fault in code that accesses the baud rate register. The most likely reason would be for m_pCan to be null but it should not be. Do you have a full sketch that demonstrates this problem?

autoprog, 2019-08-02:
Hi collin,

 running flash normals ok,

do othes test, my resume ( class constructor, startup)
see:

add: extern CANRaw myCan; in (due_can.h)

add:          in my ( main.cpp )
                myCan0.init( CAN_BPS_500K);  // call ok
                 myCan0.watchFor();                 // call ok
                ....
  Can0.sendFrame(outgoing);  // all ok, **send frame ok**.....not return (isr erro not exist )
//--------------------
it test show erro write registers in can_class ,is runtime construtor.

 I have code(library.C ) ram, canbus M4 NXP running ok
thanks,
Carlos

autoprog, 2019-08-05:
Hi Collin, thanks your help,
[ SOLVED ]
[image]

my case ram, used source.c = ok

thanks,
Carlos.