collin80/due_can #43: DUE_CAN using interrupts

skrenda, 2020-01-04:
Hello, 

I'm looking for example of code using interrupt when CAN packet is recieved.
I can not use main loop to check if data arrived.
Is possible to use your library with interrupt handling? 

Thank you

Regards Skrenda

collin80, 2020-01-04:
Yes, the library itself is always interrupt driven in the background. You can poll in your main loop but what that is really doing is checking a queue maintained by the interrupt handler. You can also directly have the interrupt send you the frame when it comes in. This is done via callbacks in the library. So, look at the various examples that use callbacks (they have CB in them). These should show you how to do it.

closed 2020-01-04 by collin80

skrenda, 2020-01-06:
Hello Colin,
 
thank you very much, I some how missed these examples. Sorry.
 
Regards Skrenda

> Od: "Collin Kidder" <notifications@github.com>
> Komu: "collin80/due_can" <due_can@noreply.github.com>
> Datum: 04.01.2020 23:35
>
> CC: <author@noreply.github.com>
Yes, the library itself is always interrupt driven in the background. You can poll in your main loop but what that is really doing is checking a queue maintained by the interrupt handler. You can also directly have the interrupt send you the frame when it comes in. This is done via callbacks in the library. So, look at the various examples that use callbacks (they have CB in them). These should show you how to do it.