collin80/due_can #44: Issue with due can library and DUE

AAKFISHERMAN, 2020-02-23:
Hi,
I'm testing canbus using DUE and transceiver sn65hvd230 shield. 
I have a canbus network as follows;

CANBUS NETWORK:
- 2 arduinos NANO with MCP2515
- 1 arduino MEGA 2560 with MCP2515
- 1 arduino DUE with sn65hvd230
> Network is property terminated with 120 ohm resistors.
> Each arduino sends at least two canbus messages, each with 8 bytes and its own ID;
> Each arduino reads canbus and shows at serial, messages received.

BEHAVIOR:
> All arduinos (NANO/MEGA) read all messages on canbus network... BUT DUE only one message... Seems other messages are ignored. But there isn't can filter configured.
> DUE sends one message, but none of arduinos can receive...

I think sketch from NANO and MEGA doesn't matter because they are working very well...

On DUE: (Simplified sketch)
#include <variant.h>
#include <due_can.h>

SETUP
void setup()
{                    
    Serial.begin(115200);
	Can0.begin(CAN_BPS_250K);
    Can0.watchFor();
}

void CANBUSTX()
{
     CAN_FRAME CANBus_TX;
     CANBus_TX.id  = 0x090;                          
     CANBus_TX.extended = false;                   
     CANBus_TX.priority = 4;                       
     CANBus_TX.length = 8;                           
     CANBus_TX.rtr = 0;
     CANBus_TX.data.byte[0] = 24;
     CANBus_TX.data.byte[1] = 33;
     CANBus_TX.data.byte[2] = 23;
     CANBus_TX.data.byte[3] = 11;
     CANBus_TX.data.byte[4] = 0;
     CANBus_TX.data.byte[5] = 0;
     CANBus_TX.data.byte[6] = 0;
     CANBus_TX.data.byte[7] = 10;
     Can0.sendFrame(CANBus_TX);
}

void CANBUSRX()
{	
  CAN_FRAME CANBus_RX;
  if (Can0.available() > 0)
  {
    Can0.read(CANBus_RX);
    Serial.print(F(">>>>>>>>>> Received CANBUS!!! "));
    Serial.println(CANBus_RX.id,HEX);
  }
}

LOOP:
void loop()
{
	CANBUSTX();
	CANBUSRX();
}

Of course i had simplified the sketch. Arduinos are sending not only constant values, but values read from sensors (byte format).

What can be wrong? With MCP2515 on arduinos and  sn65hvd230 on ESP32 (example), they works very well.

Thanks for any help
Regards
Alex

collin80, 2020-02-23:
Your example code seems correct. So, it should (in theory) work as-is. It
is possible that something in the non-simplified version changes the way it
works. But, it seems odd that it accepts one message and ignores the rest.
That sounds like a filter but you aren't filtering. You might try having
the Due just read and see if you can get that working before doing any
sending. Also, make sure nothing funny is happening with the Rs pin on the
SN65HVD230 transceiver.

AAKFISHERMAN, 2020-02-23:
Hi. 
In that sketch i'm using ethernet w5500 shiled also sdcard and ds3231 rtc.
Transceivers SN65HVD230 are shields CV230. I had tested with 3 different pieces, without success.
In fact, seems a filter applied. If i change Can0.watchFor() to  Can0.watchFor(0x222), for example, it reads frame 0x222... But only one at a time... I checked that i have 9 different canbus messages on my network. But DUE can read only one a time. I must set different mailboxes to work? Or something like that? Thanks
Alex

AAKFISHERMAN, 2020-02-25:
Hi, I did some wiring modifications on my canbus network (shielded cables) and now i´m receiving several different frames on DUE (only RX). But still not transmiting; None of arduinos receive frames from DUE, but from other arduinos OK. It´s necessary to check number os mailboxes to TX? Or something that i can check? Thanks
Alex

collin80, 2020-02-25:
The default is one TX mailbox. This should be plenty. Sometimes people
sending very rapidly like 2 TX mailboxes so that the hardware can load one
while sending the other and basically send non-stop. Still, you can do over
1000 frames per second with just one TX mailbox. I don't think that's the
issue. It's really starting to sound like the CANTX pin of the Due is not
making it to the transceiver. If you have an oscilloscope you might try a
trace of that line to make sure it is happening properly.

AAKFISHERMAN, 2020-02-26:
Hi. Thanks for your help Collin.  I did a trace with oscilloscope and found some inconsistences. Voltage on DUE CAN TX was very low.... like 0.282V... At DUE CAN RX also, 0.820V... Then i change CJMCU-230 transceiver for another one... Same problem...  Finally, i got one transceiver from an older ESP32 test board and works fine... TX and RX are working very well...  Its very strange, because transceivers i have with chipset VP230 "82M" not works... Transceivers that works are VP230 "75M" and "68M".... I don´t know differences between them. But all of VP230 82M i have (3 pieces)  not works... Thank you very much for your support.
Regadrs
Alex

alextoto51, 2020-05-04:
Hello, sorry for my English I use a translator.
I made a project with on one side a uno and a mpc2515 and on the other a Due with the sn65HDV230

I send a uno to the due I receive the id but the data is fixed while the values ​​change. I tested instead of the DUe a mega2560 + mpc2515 there it works ID + data change.

I tested live without the SN65HDV230 it's the same Id ok but fixed data!

I noticed a strange thing the uno with the mpc is set to 500Kbs and the due sees nothing in 500 but sees in 250Kbs!

I tested on CAN0 and on CAN1 it's the same!

With or without filter it's the same!
/ Arduino Due - Displays all traffic found on either canbus port
// By Thibaut Viard/Wilfredo Molina/Collin Kidder 2013-2014

// Required libraries
#include "variant.h"
#include <due_can.h>

//Leave defined if you use native port, comment if using programming port
//This sketch could provide a lot of traffic so it might be best to use the
//native port
//#define Serial SerialUSB

void setup()
{

  Serial.begin(38400);
  
  // Initialize CAN0 and CAN1, Set the proper baud rates here
  Can1.begin(CAN_BPS_250K);
  Serial.println("Doing Auto Baud scan on CAN0");
  Can1.watchFor ();
  //Can1.begin(CAN_BPS_250K);
  

  
}

void printFrame(CAN_FRAME &frame) {
   Serial.print("ID: ");
   Serial.print(frame.id);
   Serial.print(" Len: ");
   Serial.print(frame.length);
   Serial.print(" Data: ");
   for (int count = 0; count < frame.length; count++) {
       Serial.print(frame.data.bytes[count]);
       Serial.print(" ");
   }
   Serial.print("\r\n");
}

void loop(){
  CAN_FRAME incoming;

  if (Can1.available() > 0) {
	Can1.read(incoming); 
	printFrame(incoming);
  }
  //if (Can1.available() > 0) {
	//Can1.read(incoming); 
	//printFrame(incoming);
 // }
}

Help me Thanks
ALex

alextoto51, 2020-05-05:
Hi,

I come back to you, I tested with a teensy 4 and another library it did the same chose! so I tested other chosen:

If I put the UNO with mcp2515 on one side and on the other side the Mega with mcp2515 everything is ok if I put in the middle arduino DUE and your library (and another teensy4 and other library) there it works as long as it is between 2! I think of the resistance I would test.

OK:
|UNO MCP2515 Resistor------|DUE / Teensy (sn65HDV230)| --------Resistor MCP2515 Mega|

Not ok:
|UNO MCP2515 Resistor------|DUE / Teensy (sn65HDV230)|        Resistor ?
|DUE / Teensy (sn65HDV230)| --------Resistor MCP2515 Mega|   Resistor ?

If that can help another
Alex