collin80/due_can #45: Confirm receiving

bbd556, 2020-03-16:
How can i undrestand that my data frame has been received by receiver device?is there any confirmation code ?

bbd556, 2020-03-18:
Plz help me!!!

collin80, 2020-03-19:
Well, as the library sits, no. But, technically the hardware does know if the frame was successfully received by another node or not. I'd have to look into it some more but it should be possible to add such functionality to the library. There should be an interrupt that can fire when the message is successfully sent.

bbd556, 2020-04-04:
> Well, as the library sits, no. But, technically the hardware does know if the frame was successfully received by another node or not. I'd have to look into it some more but it should be possible to [...]

bbd556, 2020-04-04:
Thanks  very much mr collin for your useful library and your responsibility about users.

closed 2020-04-04 by bbd556