collin80/due_can #53: ACK bit should be 0?

spudmu, 2021-03-24:
I am trying to emulate some CAN frames on my due with my very limited C++ abilities, my background is more hardware oriented and I am very new to the due.
When I scope the CAN output of my due I see that the ACK bit is always 1 whereas on the frames I am trying to emulate the ACK bit is always 0 which I believe is correct. Is there a command in due_can to clear the ACK bit

collin80, 2021-03-24:
The ACK bit is automatically handled in hardware. If the hardware accepts a frame it asserts ACK and sends it over the wire. There really isn't any way to set that bit explicitly in the library other than to accept a frame that is received.

spudmu, 2021-03-24:
Thanks for the quick reply Collin, In that case it looks like I will not be able to emulate these frames with the due.

Jim

collin80, 2021-03-24:
You might consider looking here: https://github.com/kentindell