collin80/due_can #59: Bus speed at 500 K

Duke069, 2023-05-02:
I have no issues running my code at 125 K and all works great . The minute I run at 500 K my truck crashes .
I have all terminations done correctly and have termination resistors in  place .

I was wondering if I had to play with TQ and sample points .
My truck seems to like 75 sample rate at 125 K speed .

It works perfect at 125K but i know the timing gets more critical the faster the bus speed .

Any help would be amazing .
Thanks

Here is my basic code .

//Reads all traffic on CAN0 and forwards it to CAN1 and modifies cluster frame first.
//Code based on samples provided by Thibaut Viard/Wilfredo Molina/Collin Kidder
//------------------------------------------------------------------------------------------------

// Required libraries
#include "variant.h"
#include <due_can.h>

//Leave defined if you use native port, comment if using programming port
//This sketch could provide a lot of traffic so it might be best to use the
//native port
#define Serial SerialUSB

 void printFrame(CAN_FRAME *frame, int filter) {
   Serial.print("Fltr: ");
   if (filter > -1) Serial.print(filter);
   else Serial.print("???");
   Serial.print(" ID: 0x");
   Serial.print(frame->id, HEX);
   Serial.print(" Len: ");
   Serial.print(frame->length);
   Serial.print(" Data: 0x");
   for (int count = 0; count < frame->length; count++) {
       Serial.print(frame->data.bytes[count], HEX);
       Serial.print(" ");
   }
   Serial.print("\r\n");
}

void gotFrame0(CAN_FRAME *frame) 
{
   //printFrame(frame, 1);  //uncomment line to print frames that are going out
   Can1.sendFrame(*frame);
}

void gotFrameCluster(CAN_FRAME *frame) //cluster
{
   frame->data.byte[1]=0x08;
   frame->data.byte[6]=0x01;
   //printFrame(frame, 2);  //uncomment line to print frames that are going out
   Can1.sendFrame(*frame);
}

void gotFrame1(CAN_FRAME *frame) 
{
   //printFrame(frame, -1);  //uncomment line to print frames that are going out
   Can0.sendFrame(*frame);
}

void setup()
{

  //Serial.begin(250000);  //Uncomment for serial
  
  // Initialize CAN0, Set the proper baud rates here
  Can0.begin(CAN_BPS_500K);
  // Initialize CAN1, Set the proper baud rates here
  Can1.begin(CAN_BPS_500K);
  
  //By default there are 7 RX mailboxes for each device - Standard frames
  //Can0 Filters
  Can0.setRXFilter(0, 0x372, 0x7FF, false); //Cluster
  
  //Can1 Filters
  Can1.setRXFilter(0, 0, false); //catch all mailbox - no mailbox ID specified

  //now register all of the callback functions.
  Can0.setCallback(0, gotFrameCluster);
 
  //this function will get a callback for any mailbox that doesn't have a registered callback
  Can0.setGeneralCallback(gotFrame0);
  Can1.setGeneralCallback(gotFrame1);
}

void loop(){ //note the empty loop here. All work is done via callback as frames come in - no need to poll for them
}

collin80, 2023-07-25:
The sketch seems fine. You seem to have covered your bases on how you've attempted to fix the problem. But, you're right, there could be problems with the sampling point. You can see in due_can.h that most sampling points are set around 65-75%. Sometimes upwards of 87% is best. Probably less than 65% isn't the best idea. But, usually the sampling point only has to be increased if you have really long bus wires. On vehicle scale wiring it shouldn't be so critical. But, there is another thing I suspect. You say it's properly terminated but sometimes things get over terminated and that can cause issues too. Make sure you aren't adding termination where it should not be. If you are doing MitM then check the resistance of the wiring on both sides. If it is already 60 ohms then don't add any more. If it is 120 ohms then go ahead and add a 120 ohm resistor on that side.

Duke069, 2023-07-25:
Thanks for your feedback and support on this . 
I am using the copperhill tech due can board with the two channel can transceivers .So for can 0 which is truck side I have removed the eol jumper so this uses the two star can blocks on the truck which is factory configuration . 
Can1 goes to my instrument cluster and I have the eol jumper on and then I have another star can block for the eol .This gives me 60 ohms on can0 and 60 ohms on can 1. 
I have my can 1 star can block grounded as well just like the factory does . 
I changed the TQ min and Max to 13 and used the sample point at 75 percent which didn't solve the problem . 

I also tried not connecting can 1 to the truck and just have can 0 connected to the truck and that also crashed the truck . 
I am really stumped on this one as at 125 K it works really well using all 7 mailboxes as well . 

This is the board i am using . 
I was questing maybe the transceivers on this guy . 
I have four of these but all of them behave the same way . 

Thanks again for taking the time to help me with this .

Duke069, 2023-07-25:
I have added my friend Roman who has been a huge help for me with canbus . 
Thanks again

Duke069, 2023-07-25:
I verified with an ohm meter just to make sure the resistance is correct and I am getting 60 or so ohms per can channel .
Polarity is correct for can H and can L on both busses .
Thanks again Collin for your time and expertise .
Always appreciate these gourds and efforts people put in to help others !
Cheers

collin80, 2023-07-25:
It sounds like you've done everything correctly yet it still isn't working. There are a couple more things I can think to try:

1. Put the CAN ports into listen only mode. This should case your board to not do anything to the CAN bus. If you can run the vehicle stock with the board attached and it doesn't fault then that's a good sign and might point to what to try next. If it faults even in listen only mode then something is wrong at a hardware level - termination (though you checked that), ground loop, etc.
2. Uncomment the printFrame lines so you can see when traffic comes in. What you're hoping to see is that traffic is actually flowing in. Never seeing any traffic is a bad sign.

I guess it'd be possible that your CAN capture board is strange in some way or the transceiver is broken but you would certainly think that 4 would not all be bad and you'd also think that 125k wouldn't work then either.

Duke069, 2023-07-26:
Thanks Collin I will do some playing this weekend with it and give it another solid go .
Thanks for all your help and I will keep you posted on my findings .
Cheers !

Duke069, 2023-07-27:
Well I updated to IDE 2.1 and now it's working good 
You can see me logging on can 1 and the code is working great .
[image]![Uploading 256652475...]

Collin thanks for all your help with this !
So glad it's working now !

Duke069, 2023-07-27:
[image]
Can logger showing the code is working

Duke069, 2024-04-02:
Good day sir I hope you are doing well. I was hoping to pick your brain a little bit if possible. 
I am trying to add a delay50 write before can0 send to one of my can messages but when i add the delay it causes my code to stop working all together. 
I am sure i am doing something wrong but any insight would be most appreciated. 
Here is my code. 
Do i need to uncomment include < TimerControl.h in the constructor ? 

//#include <DueCANLayer.h> 
//#include <TimerControl.h> 
//#include <can_common.h> 
//#include <due_can.h> 

// Arduino Due - Displays all traffic found on either canbus port 
//Modified from the more generic TrafficSniffer sketch to instead use 
//callback functions to receive the frames. Illustrates how to use 
//the per-mailbox and general callback functionality 
// By Thibaut Viard/Wilfredo Molina/Collin Kidder 2013-2014 

// Required libraries 
#include "variant.h" 
#include <due_can.h> 

//Leave defined if you use native port, comment if using programming port 
//This sketch could provide a lot of traffic so it might be best to use the 
//native port 
#define Serial SerialUSB 

void printFrame ( CAN_FRAME * frame , int filter) { 
Serial . print ( "Fltr: " ) ; 
if ( filter > - 1 ) Serial . print ( filter ) ; 
else Serial . print ( "???" ) ; 
Serial . print ( " ID: 0x" ) ; 
Serial . print ( frame -> id , HEX ) ; 
Serial . print ( " Len: " ) ; 
Serial . print ( frame -> length ) ; 
Serial . print ( " Data: 0x" ) ; 
for ( int count = 0 ; count < frame -> length ; count++ ) { 
Serial . print ( frame -> data . bytes [count], HEX ) ; 
Serial . print ( " " ) ; 
} 
Serial . print ( "\r\n" ) ; 

} 
void gotFrame0 ( CAN_FRAME * frame) 
{ 
} 

void gotFrameMB4 ( CAN_FRAME * frame) //Config 7 
{ 
frame -> data . byte [ 0 ] = 0x 81 ; //"Programmed" Status 
//printFrame(frame, 2); //uncomment line to print frames that are going out 
Can0 . sendFrame ( *frame ) ; //Forward on the VehConfig7 Frame 
//Spoof VehConfig8 Frame 
frame -> id = 0x 38A ; 
frame -> data . byte [ 0 ] = 0x 01 ; //"Programmed" Status 
frame -> data . byte [ 1 ] = 0x 00 ; 
frame -> data . byte [ 2 ] = 0x 80 ; //0x00 Off road camera off 
frame -> data . byte [ 3 ] = 0x 82 ; //0x00 Digital CHMSL Off 02 on 
frame -> data . byte [ 4 ] = 0x 21 ; //Trailer Light check 
frame -> data . byte [ 5 ] = 0x 01 ; //0x00 Aux trailer camera off 
frame -> data . byte [ 6 ] = 0x A9 ; //0x40 Trailer reverse guidance off//59/79/B No tach type enabled/A9 Default 
frame -> data . byte [ 7 ] = 0x 25 ; 
//Can0.sendFrame(*frame); 
} 

void gotFrameMB5 ( CAN_FRAME * frame) //CSM2 
{ 
//printFrame(frame, 2); //uncomment line to print frames that are going out 
frame -> data . byte [ 6 ]=0x 60 ; //power side step csm2 
Can0 . sendFrame ( *frame ) ; //Forward on the CSM2 Frame 
//Spoof CSM3 Frame 
[... 151 lines cut ...]
// Initialize CAN1, Set the proper baud rates here 
//Can1.begin(CAN_BPS_500K); 
//standard 
Can0 . setRXFilter ( 4 , 0x 3E8 ,0x 7FF , false ) ; //Config7/Spoof Config 8 
Can0 . setRXFilter ( 5 , 0x 3B4 ,0x 7FF , false ) ; //CSM2/Spoof CSM3 
Can0 . setRXFilter ( 6 , 0x 318 ,0x 7FF , false ) ; //Button Mapping 
Can0 . setRXFilter ( 7 , 0x 318 ,0x 7FF , false ) ; // Heart Beat 
Can0 . setRXFilter ( 8 , 0x 3f3 ,0x 7FF , false ) ; 
// Can0.setRXFilter(0, 0x318,0x7FF,false); //catch all mailbox - no mailbox ID specified 
//Can1 Filters 
// Can1.setRXFilter(0, 0, false); //catch all mailbox - no mailbox ID specified 
//now register all of the callback functions. 
//Can0.setCallback(0, gotFrameMB0); 
Can0 . setCallback ( 4 , gotFrameMB4 ) ; //Config7/Spoof Config8 
Can0 . setCallback ( 5 , gotFrameMB5 ) ; //CSM2//Spoof CSM3 
Can0 . setCallback ( 6 , gotFrameMB6 ) ; // Steering Buttons 
Can0 . setCallback ( 7 , gotFrameMB7 ) ; // Heart Beat 
Can0 . setCallback ( 8 , gotFrameMB8 ) ; // 3f3 
//this function will get a callback for any mailbox that doesn't have a registered callback from above -> 2 and 6 
Can0 . setGeneralCallback ( gotFrame0 ) ; 
//Can1.setGeneralCallback(gotFrameMB1); 
} 

void loop (){ //note the empty loop here. All work is done via callback as frames come in - no need to poll for them 
}

Duke069, 2024-04-12:
Good day sir!I tried a few other things but I can’t seem to get the delay to work as my due just locks up when I add the delay.I’m willing to try any suggestions at this point.Thanks in advance for any help!Cheers.Sent from my iPhone

Put the CAN ports into listen only mode. This should case your board to not do anything to the CAN bus. If you can run the vehicle stock with the board attached and it doesn't fault then that's a good sign and might point to what to try next. If it faults even in listen only mode then something is wrong at a hardware level - termination (though you checked that), ground loop, etc.Uncomment the printFrame lines so you can see when traffic comes in. What you're hoping to see is that traffic is actually flowing in. Never seeing any traffic is a bad sign.

edmsilva80, 2025-01-16:
Hi,

I am trying to use the DUE_CAN library with a 500K baudrate but I am getting the CAN bus fill with "raw" data.

I am using Arduino DUE with a MCP2551 board as data emitter and as receiver also Arduino DUE with a different CAN board.
CAN bus with a 60ohm resistance (2x 120ohm).
Programming tool used is Arduino DUE v2.3.4.

The receiver, Arduino DUE with a CAN controller + transceiver (using SPI) set to receive at 500K as no issue.
Tested with other boards (emitters), and works when the emitter that I mentioned is set to 250K.

I analysed the CAN bus (CAN_H; CAN_L) directly with an oscilloscope and see the garbage in the line (there is no period where is no data), the bus is always occupied.

My question is, can there be an issue with the library when the baudrate is different from 250K?
I have tried 500K and 125K, the result is the same.

I think we can remove the MCP2551 as source of the problem, since it works at 250K, and looking to the CANTX directly on the Arduino DUE (emitter) the line is also always busy.

I see that changing the  sampling points higher than 75% could help, but since the problem can in the emitter side, should this be changed?